pyMultiVideo/code #10: GPIO pinstate may not correspond to correct images

ThomasAkam, 2025-02-08:
The length of sync pulses as recorded in the GPIO pinstate data file appears more variable than they should be.  Looking at [SpinnakerCamera.retrieve_buffered_data](https://github.com/pyMultiVideo/code/blob/main/camera_api/spinnaker.py#L314C9-L314C31) it appears that there is no guarentee that the GPIO pin states returned correspond to the image frames.  This is because the GPIO pin states are not recovered from the images in a one to one fashion, but rather are read from the camera directly in [SpinnakerCamera.get_GPIO_data](https://github.com/pyMultiVideo/code/blob/main/camera_api/spinnaker.py#L383).  I think the correct way to get the pinstates corresponding to each image would be something more like:

```python
next_image = self.cam.GetNextImage(0) # Get image object from camera
image_pixels = next_image.GetNDArray()  # Extract the  image pixels as numpy array.
chunk_data = next_image.GetChunkData() # Get the additional data chunk containing the timestamp and GPIO pinstates
image_timestamp = chunk_data.GetTimestamp() # Get timestamp from the chunk data.
line_status = chunk_data.GetLineStatusAll() # Get GPIO line status from chunk data.
next_image.Release() # Release image after accessing chunk data.
```

See also the ChunkData [documentation](https://softwareservices.flir.com/Spinnaker/latest/class_spinnaker_1_1_chunk_data.html).

ThomasAkam, 2025-02-12:
@alifuaziz I think I may have worked out why we are not able to get GPIO pinstates from images ChunkData.  Based on looking at [this](https://github.com/jwang945/PySpinDystonia/blob/main/Examples/Python2/ChunkData.py) code example, I think that you have to call ChunkEnable once for each type of chunk data you want to enable, after selecting the relevant item using ChunkSelector, so e.g. to enable both the timestamp and GPIO pinstates you would do something like:

```
cam.ChunkSelector.SetValue(PySpin.ChunkSelector_Timestamp)
cam.ChunkEnable.SetValue(True)
cam.ChunkSelector.SetValue(PySpin.ChunkSelector_LineStatusAll )
cam.ChunkEnable.SetValue(True)
```

alifuaziz, 2025-02-14:
I have done a long term test of the GPIO state acquisition can compared it to the pyControl task's report of the time GPIO state has changed from on to off. 

I have used to the time of file create to act as the start time for the pyMultiVideo timestamps  (`2025-02-14 11:57:41.538587`). The pyControl task reports the start time to milliseconds resolution.

[GPIO-TEST-3_2025-02-14_11-57-41_GPIO_data.csv](https://github.com/user-attachments/files/18800225/GPIO-TEST-3_2025-02-14_11-57-41_GPIO_data.csv)

[GPIO-TEST-3-2025-02-14-115747.csv](https://github.com/user-attachments/files/18800253/GPIO-TEST-3-2025-02-14-115747.csv)

I have looked into what the time differences are between the two different applications' report of when the sync pulses are changing. 
The graph of the differences in their reported times is as follows: 

[image]

- It looks like there is a change in the differences over time. (to begin with the pyMultiVideo reports the change before pyControl's clock is running faster than spinnaker clock.

ThomasAkam, 2025-02-14:
> I have done a long term test of the GPIO state acquisition can compared it to the pyControl task's report of the time GPIO state has changed from on to off.
>
> I have used to the time of file create [...]

I don't think this is an issue - we expect the camera clock to drift a bit relative to the computer time which is why we use the GPIO pins for synchronisation.

ThomasAkam, 2025-02-14:
This issue is now resolved in spinnaker_refactor branch at commit 0ad18247c742ee83d61097f895d3afed72b7de28 by getting GPIO states for each aquired image directly from the image data.  This functionallity is not implemented in spinnaker chunkdata so had to use a workaround provided by FLIR support.

closed 2025-02-14 by ThomasAkam