pyMultiVideo/code #17: Frame rate not being set to manual

ThomasAkam, 2025-02-13:
When using with a new camera the frame rate was not being set correctly and only started working after I set the frame rate control to manual in SpinView.

alifuaziz, 2025-02-14:
```
import PySpin

SERIAL_NUMBER = '21013409'

system = PySpin.System.GetInstance()
cam = system.GetCameras().GetBySerial(SERIAL_NUMBER)
cam.Init()
nodemap = cam.GetNodeMap()

# Raises Error
fra_node = PySpin.CEnumerationPtr(nodemap.GetNode("AcquisitionFrameRateAuto"))
fra_node.SetIntValue(fra_node.GetEntryByName("Off").GetValue())
frc_node = PySpin.CBooleanPtr(nodemap.GetNode("AcquisitionFrameRateEnabled"))
frc_node.SetValue(True)

# Does not raise error
node_line_status_all = PySpin.CIntegerPtr(nodemap.GetNode('LineStatusAll'))
if PySpin.IsAvailable(node_line_status_all) and PySpin.IsReadable(node_line_status_all):
    entry_node = node_line_status_all.GetValue()
    print(entry_node)
```
Testing the camera that didnt work: setting the framerate node raised an error. Getting the linestatus seemed to work fine. 

This was when running the code in isolation. 

When running as part of pmv, this was not the case and both trying to get the lineststatus raised an error and trying to set the framerate raised an error: 

```
Camera: <PySpin.PySpin.CameraPtr; proxy of <Swig Object of type 'Spinnaker::CameraPtr *' at 0x0000018D1FC2D8C0> >
Error Setting Frame Rate: Spinnaker: GenICam::AccessException= Node is not writable. : AccessException thrown in node 'AcquisitionFrameRate' while calling 'AcquisitionFrameRate.SetValue()' (file 'FloatT.h', line 83) [-2006]
Traceback (most recent call last):
  File "C:\Users\alifa\Documents\pymv_testing\GUI\viewfinder_widget.py", line 505, in change_camera
    self._change_camera(camera_unique_id, new_camera_label)
  File "C:\Users\alifa\Documents\pymv_testing\GUI\viewfinder_widget.py", line 514, in _change_camera
    self.camera_api = init_camera_api(new_unique_id, self.camera_settings)
  File "C:\Users\alifa\Documents\pymv_testing\tools\load_camera.py", line 141, in init_camera_api
    return camera_module.initialise_by_id(
  File "C:\Users\alifa\Documents\pymv_testing\camera_api\spinnaker.py", line 232, in initialise_by_id
    return SpinnakerCamera(unique_id=_id, CameraConfig=CameraSettingsConfig)
  File "C:\Users\alifa\Documents\pymv_testing\camera_api\spinnaker.py", line 46, in __init__
    fra_node.SetIntValue(fra_node.GetEntryByName("Off").GetValue())
  File "C:\miniconda3\envs\pyMultiCam_env\lib\site-packages\PySpin\PySpin.py", line 12844, in SetIntValue
    return _PySpin.CEnumerationPtr_SetIntValue(self, Value, Verify)
_PySpin.SpinnakerException: Spinnaker: GenICam::AccessException= Node is not writable. : AccessException thrown in node 'AcquisitionFrameRateAuto' while calling 'AcquisitionFrameRateAuto.SetIntValue()' (file 'EnumerationT.h', line 83) [-2006]
Traceback (most recent call last):
  File "C:\Users\alifa\Documents\pymv_testing\GUI\video_capture_tab.py", line 234, in fetch_image_data
    camera_widget.fetch_image_data()
  File "C:\Users\alifa\Documents\pymv_testing\GUI\viewfinder_widget.py", line 192, in fetch_image_data
    new_images = self.camera_api.get_available_images()
  File "C:\Users\alifa\Documents\pymv_testing\camera_api\spinnaker.py", line 161, in get_available_images
    line_status = PySpin.CIntegerPtr(self.nodemap.GetNode("LineStatusAll")).GetValue()
  File "C:\miniconda3\envs\pyMultiCam_env\lib\site-packages\PySpin\PySpin.py", line 11583, in GetValue
    return _PySpin.CIntegerPtr_GetValue(self, Verify, IgnoreCache)
_PySpin.SpinnakerException: Spinnaker: GenICam::AccessException= Node is not readable. : AccessException thrown in node 'LineStatusAll' while calling 'LineStatusAll.GetValue()' (file 'IntegerT.h', line 149) [-2006]
```

alifuaziz, 2025-02-16:
When chaning the frame rate from the camera settings tab, it doesn't update when the camera restarts when you go back to the viewfinder tab. This is because the camera doesn't get asked to start running with the camera settings config from the `on_tab_change` function. This should be added to fix this bug.

closed 2025-02-16 by alifuaziz